ethz-asl/versavis #5: the basler camera ros-driver failed

minxuanjun, 2019-12-17:


minxuanjun, 2019-12-19:
@floriantschopp   Hi, thanks for your open source, it's really an exciting solution for VIO harware hardware synchronization! I test currently your program with basler acA1920-155uc camera. but I find that the basler camera ros driver link had failed. If  you have suitable basler camera ros driver, can you share me?
Thank you!

floriantschopp, 2019-12-19:
Hi @minxuanjun 
Unfortunately, currently I cannot open-source the version of the basler camera ros driver that I am using. However, the changes needed for VersaVIS to work are small, mainly the publishing of the `image_numbered` message and maybe activation of trigger and exposure output mode. 
Check the changes made to the [flir_camera_driver](https://github.com/ethz-asl/flir_camera_driver/pull/10) for a reference, mainly:
* [Advertising new message](https://github.com/ethz-asl/flir_camera_driver/blob/fe169349da18fa702021d83a5b354e3edd7249c1/spinnaker_camera_driver/src/nodelet.cpp#L350)
* [Setting properties of new message](https://github.com/ethz-asl/flir_camera_driver/blob/fe169349da18fa702021d83a5b354e3edd7249c1/spinnaker_camera_driver/src/nodelet.cpp#L634)

closed 2020-03-04 by floriantschopp