ethz-asl/versavis #13: RGB IMU calibration

amirtakouh, 2020-03-11:
Hi,

I have been trying to calibrate an RGB and IMU sensor using Kalibr. I am using the same camera and IMU as you do in your project. Camera is BFS-U3-04S2M and I believe the lens is also the same as yours (as much as I can say from the image of your setup in the paper). The problem is that my reprojection errors are quite higher than what you have achieved in the paper.
Could you please let me know what kind of settings you are using in the camera and if there is something about the lens like getting the right focus, etc?
Attached, please find the Kalibr report and also a sample image of the camera.
[image]
[2020_03_10_17_33_19_481.pdf](https://github.com/ethz-asl/versavis/files/4319344/2020_03_10_17_33_19_481.pdf)

floriantschopp, 2020-03-12:
Hi amirtakouh,

Just to make sure, are you using a color or monochrome camera (BFS-U3-04S2M is a monochrome camera ;-))? 
Looking at your report, it seems like your IMU noise parameters are pretty high, what IMU do you use? 
To see whether problems at estimating the camera intrinsics are influencing the result, it would be interesting to see the report of the camera intrinsics calibration. Also, a strong influence has time synchronization between camera and IMU, how are you synchronizing them?
Last but not least, it is crucial to have a high-quality target. If I can see it correctly on the attached image, it looks like you have a printed April tag target. You can highly improve the results by making sure that the target parameters are very accurate and that the calibration target is very flat and has no bumps etc.
For camera focus, use a [Siemens Star](https://en.wikipedia.org/wiki/Siemens_star) and try to make the blurry region in the middle as small as possible when keeping your camera at your working distance.

I hope this helps.

amirtakouh, 2020-03-12:
Hi,

Thank you for the quick response and good information!
Sorry for the miss-information. The camera model is actually BFS-U3-04S2C-CS which is a color camera. However it is being used in Mono8 mode. Does this affect the re-projection error?

The IMU is from Analog Devices, ADIS16448. I guess you have used the same in your setup. Are there any specific settings for the IMU?

The IMU and camera are synchronized in a microcontroller. Both IMU and camera trigger signal are connected to the microcontroller. As soon as the DATA_READY pin of the IMU is activated, the microcontroller increments a counter. Whenever the counter reaches a certain number (8), the microcontroller triggers the camera and counter is reset. So basically the camera is triggered for every 8 IMU samples.

floriantschopp, 2020-03-12:
OK! As far as I know, color cameras are slightly harder to calibrate due to chromatic aberration but the effect will not be as high. 
Yes, that's the same IMU we are using with different filtering settings (see paper). Your synchronization scheme should work in my opinion if you have a constant exposure time (you should try with small fixed exposure times like 1ms), otherwise, you might get problems with exposure time compensation. Furthermore, you have to make sure that the measurements are actually triggered periodically (are you using hardware timers?).

The remaining errors I could imagine are:
* Calibration target settings are wrong
* Calibration target is not flat (might be my best guess atm)
* IMU noise parameters are can have an influence, you can try with the following ones
```
rostopic: your/topic
update_rate: 200.0 #Hz

accelerometer_noise_density: 0.01 # continous
accelerometer_random_walk: 0.0002
gyroscope_noise_density: 0.005 #continous
gyroscope_random_walk: 4.0e-06

```

floriantschopp, 2020-07-07:
Closed due to inactivity.

closed 2020-07-07 by floriantschopp