ethz-asl/versavis #17: host computer time synchronization

CrazyPopLin, 2020-10-20:
Hi, thanks for sharing this awesome project. I have a question about host computer time synchronization. 

In your paper Fig.2, it's seems the host computer is time sync with MCU. Also, in page 8,  Ouster OS-1 64-beam LiDAR is time sync with PTP by host computer. All of this seems to say, host computer is time synchronized. (maybe be using gpsd, pps and fake GPS time). 

Could you confirm it !! 

I am working on time sync for multi-sensor project. But my sonar can't do time sync by external trigger or PPS. Only to software trigger, which means my host computer has to time sync with MCU which used time sync for cameras and IMU. I know host computer can be time sync with gps and PPS, but not sure with fake gps message and PPS.

Thanks for your help.

floriantschopp, 2021-03-10:
Hi @linzha0 
Sorry for my late answer. It's actually the other way around, the MCU is time synchronized with the host which acts as the clock master. Maybe Fig.2 is a bit misleading in that case, sorry for that. Also, the Ouster OS-1 we use is time-synchronized to the host (with the host as master). In parallel, you could also time sync the host to GPS time in case you have GPS reception.

For your project, this is very similar to the use case when you have an odometry sensor that is time-stamped on the host computer. Thanks to the MCU performing time sync to the host, the timestamps obtained on both systems (host vs MCU) should be comparable and can be considered to be (software) synchronized.

Hope this helps!
Best, Flo

closed 2021-03-10 by floriantschopp

linzha0, 2021-03-31:
Hi @floriantschopp , thanks for your reply. May I ask how to time sync the MCU from host computer?

floriantschopp, 2021-04-01:
Hi @linzha0 
This happens automatically through the rosserial link (using the EKF to track clock offset and skew).
Best, Flo