ethz-asl/versavis #22: Unable to build firmware

Jaeyoung-Lim, 2021-03-31:
I was following the instructions, but was unable to build the firmware through Arduino IDE (1.8.2)

```
Arduino: 1.8.2 (Linux), Board: "VersaVIS"

Build options changed, rebuilding all
In file included from /home/jaeyoung/catkin_ws/src/versavis/firmware/libraries/versavis/src/Imu.h:17:0,
                 from /home/jaeyoung/catkin_ws/src/versavis/firmware/libraries/versavis/src/ADIS16448BMLZ.h:33,
                 from /home/jaeyoung/catkin_ws/src/versavis/firmware/versavis/versavis.ino:15:
/home/jaeyoung/catkin_ws/src/versavis/firmware/libraries/versavis/src/Sensor.h:19:10: fatal error: versavis/ImuMicro.h: No such file or directory
 #include <versavis/ImuMicro.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board VersaVIS.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

Jaeyoung-Lim, 2021-04-01:
Thanks @floriantschopp for helping me figuring out the problem!

I was doing two things wrong
a. I cloned the repository without the submodules. Solved by 
```
git submodule udpate --init --recursive
```
b. I had `ros-melodic-rosserial-arduino` installed. This should be removed and the `rosserial` in the submodule should be used instead.

c. I had a problem in the `setup.sh` script. `bash` was not being detected from https://github.com/ethz-asl/versavis/blob/8a6f5b47c13dc9dc047c23114b417194ff6a5b9e/firmware/setup.sh#L2

After the above changes, all looked great.

closed 2021-04-01 by Jaeyoung-Lim

zhangyuan12121, 2021-04-27:
How to solve the problem of missing ImuMicro.h file？

PR #30 (2021-04-28): change method to detect current shell
necessary to fix #22 and #28