ethz-asl/versavis #25: Versavis seems to crash when using with ADIS16448AMLZ

Jaeyoung-Lim, 2021-04-05:
**Problem Description**
Versavis seems to lose connection when running with the ADIS16448AMLZ IMU

Log as the following
```
$ roslaunch versavis run_versavis.launch 
... logging to /home/jaeyoung/.ros/log/64612cb0-9607-11eb-afac-002315d18521/roslaunch-jaeyoung-ThinkPad-T480-17716.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jaeyoung-ThinkPad-T480:34799/

SUMMARY
========

PARAMETERS
 * /BFS_camera_manager/num_worker_threads: 4
 * /BFS_usb_0/camera_nodelet/acquisition_frame_rate_enable: False
 * /BFS_usb_0/camera_nodelet/acquisition_mode: Continuous
 * /BFS_usb_0/camera_nodelet/auto_exposure_time_lower_limit: 300
 * /BFS_usb_0/camera_nodelet/auto_exposure_time_upper_limit: 5000
 * /BFS_usb_0/camera_nodelet/auto_gain: Continuous
 * /BFS_usb_0/camera_nodelet/auto_white_balance: Continuous
 * /BFS_usb_0/camera_nodelet/enable_trigger: On
 * /BFS_usb_0/camera_nodelet/exposure_auto: Continuous
 * /BFS_usb_0/camera_nodelet/exposure_mode: Timed
 * /BFS_usb_0/camera_nodelet/frame_id: cam0
 * /BFS_usb_0/camera_nodelet/line_inverter: False
 * /BFS_usb_0/camera_nodelet/line_mode: Output
 * /BFS_usb_0/camera_nodelet/line_selector: Line2
 * /BFS_usb_0/camera_nodelet/line_source: ExposureActive
 * /BFS_usb_0/camera_nodelet/serial: 19128740
 * /BFS_usb_0/camera_nodelet/trigger_activation_mode: RisingEdge
 * /BFS_usb_0/camera_nodelet/trigger_source: Line0
 * /BFS_usb_0/compile_nodelet_cam0/driver_topic: /BFS_usb_0/image_...
 * /BFS_usb_0/compile_nodelet_cam0/imu_offset_us: 0
 * /BFS_usb_0/compile_nodelet_cam0/versavis_topic: /versavis/cam0/
 * /BFS_usb_1/camera_nodelet/acquisition_frame_rate_enable: False
 * /BFS_usb_1/camera_nodelet/acquisition_mode: Continuous
 * /BFS_usb_1/camera_nodelet/auto_exposure_time_lower_limit: 300
 * /BFS_usb_1/camera_nodelet/auto_exposure_time_upper_limit: 5000
 * /BFS_usb_1/camera_nodelet/auto_gain: Continuous
 * /BFS_usb_1/camera_nodelet/auto_white_balance: Continuous
 * /BFS_usb_1/camera_nodelet/enable_trigger: On
 * /BFS_usb_1/camera_nodelet/exposure_auto: Continuous
 * /BFS_usb_1/camera_nodelet/exposure_mode: Timed
 * /BFS_usb_1/camera_nodelet/frame_id: cam1
 * /BFS_usb_1/camera_nodelet/line_inverter: False
 * /BFS_usb_1/camera_nodelet/line_mode: Output
 * /BFS_usb_1/camera_nodelet/line_selector: Line2
 * /BFS_usb_1/camera_nodelet/line_source: ExposureActive
 * /BFS_usb_1/camera_nodelet/serial: 18286011
 * /BFS_usb_1/camera_nodelet/trigger_activation_mode: RisingEdge
 * /BFS_usb_1/camera_nodelet/trigger_source: Line0
 * /BFS_usb_1/compile_nodelet_cam1/camera_info_topic: 
 * /BFS_usb_1/compile_nodelet_cam1/driver_topic: /BFS_usb_1/image_...
 * /BFS_usb_1/compile_nodelet_cam1/imu_offset_us: 0
 * /BFS_usb_1/compile_nodelet_cam1/publish_every_n_image: 10
 * /BFS_usb_1/compile_nodelet_cam1/publish_slow_images: False
 * /BFS_usb_1/compile_nodelet_cam1/versavis_topic: /versavis/cam1/
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /versavis_imu_receiver/imu_acceleration_covariance: 0.043864908
 * /versavis_imu_receiver/imu_accelerator_sensitivity: 0.000833
 * /versavis_imu_receiver/imu_gyro_covariance: 6e-9
 * /versavis_imu_receiver/imu_gyro_sensitivity: 0.04
 * /versavis_imu_receiver/imu_sub_topic: /versavis/imu_micro

NODES
[... 63 lines cut ...]
[WARN] [1617624410.018550]: Last read step: data checksum
[WARN] [1617624410.022428]: Run loop error: [Errno 5] Input/output error
Traceback (most recent call last):
  File "/home/jaeyoung/catkin_ws/src/versavis/dependencies/rosserial/rosserial_python/nodes/serial_node.py", line 89, in <module>
    client.run()
  File "/home/jaeyoung/catkin_ws/src/versavis/dependencies/rosserial/rosserial_python/src/rosserial_python/SerialClient.py", line 554, in run
    self.port.flushInput()
  File "/usr/local/lib/python2.7/dist-packages/serial/serialutil.py", line 588, in flushInput
    self.reset_input_buffer()
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 683, in reset_input_buffer
    self._reset_input_buffer()
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 677, in _reset_input_buffer
    termios.tcflush(self.fd, termios.TCIFLUSH)
termios.error: (5, 'Input/output error')
[rosserial_python-9] process has died [pid 17775, exit code 1, cmd /home/jaeyoung/catkin_ws/src/versavis/dependencies/rosserial/rosserial_python/nodes/serial_node.py _port:=/dev/versavis _baud:=250000 __name:=rosserial_python __log:=/home/jaeyoung/.ros/log/64612cb0-9607-11eb-afac-002315d18521/rosserial_python-9.log].
log file: /home/jaeyoung/.ros/log/64612cb0-9607-11eb-afac-002315d18521/rosserial_python-9*.log
[resetter-10] process has finished cleanly
log file: /home/jaeyoung/.ros/log/64612cb0-9607-11eb-afac-002315d18521/resetter-10*.log
[rosserial_python-9] restarting process
process[rosserial_python-9]: started with pid [17872]
[INFO] [1617624410.471204]: ROS Serial Python Node
[INFO] [1617624410.482274]: Connecting to /dev/versavis at 250000 baud
[ERROR] [1617624410.486844]: Error opening serial: [Errno 2] could not open port /dev/versavis: [Errno 2] No such file or directory: '/dev/versavis'

```

floriantschopp, 2021-04-06:
This first restart of the VersaVIS board is intended as it resets all initialization from potential previous runs. This should only happen once in the beginning when launching `run_versavis.launch`.

Also, did you change your IMU to `USE_ADIS16448AMLZ` [here](https://github.com/ethz-asl/versavis/blob/8a6f5b47c13dc9dc047c23114b417194ff6a5b9e/firmware/libraries/versavis/src/versavis_configuration.h#L39)?

Jaeyoung-Lim, 2021-04-06:
@floriantschopp I did change it in the configuration files, and I think indeed this only happens once.

Closing since this seems to be the desired behavior!

closed 2021-04-06 by Jaeyoung-Lim