ethz-asl/versavis #26: plus generated by 1hz is not stable

linzha0, 2021-04-09:
Hi, this question actually not directly to your project. If you have time, please check it.

Because my microstrain IMU only have PPS time sync option. I am using your project to generate 1Hz PPS, I use the code like camera does. I am using Arduino Zero and tried pin 16 and pin 13 as the output separately. I tried all the timer, they all have same issues.

**The issue:**
When I disable the actually trigger(like:Sensor::trigger(trigger_pin_, TRIGGER_PULSE_US*10000, type_)), the timer is stable interrupt every 1s. But when I enable the trigger to generate 100ms High and 900ms Low PPS. I record the time when the trigger is processed:

> stamp: 
>   secs: 1617982338
>   nsecs: 583847082
> ---
> stamp: 
>   secs: 1617982339
>   nsecs: 484847082
> ---
> stamp: 
>   secs: 1617982340
>   nsecs: 385847082
> ---
> stamp: 
>   secs: 1617982341
>   nsecs: 583608999
> ---
> stamp: 
>   secs: 1617982342
>   nsecs: 484608999
> ---
> stamp: 
>   secs: 1617982343
>   nsecs: 385608999
 
So the timer is seems like interrupt as 900ms, 900ms, 1200ms, 900ms, 900ms... It's definitely related to my generated plus which is about 100ms. 

**My questions:** Is this normal performance? Timer is just not accuracy enough.  Or I didn't implement the code right? Could you give me some suggestions. Thanks for your time.

floriantschopp, 2021-04-12:
Hi @linzha0 

Hmm that looks weird...
Could you post a minimal example of `.ino` code you are using?

Best, Flo

linzha0, 2021-04-12:
Hi @floriantschopp , Here is the [code](https://drive.google.com/drive/folders/1IaX5Vo4eWcRAhgD405eJzEc5BPKXzTgV?usp=sharing). Thanks for your time.

floriantschopp, 2021-04-13:
I can't see anything obviously wrong at the moment. Did you try the code as it is on Google Drive (with commented `Pps.cpp/#L69`)? With that part commented, there should definitely be a constant period...The thing you experience is definitely not the normal performance I would expect. The timers are typically accurate up to <1ms!

floriantschopp, 2021-04-13:
Also you don't really need the line `Pps.cpp/#L74` as your compare time is always the same (1s)

linzha0, 2021-04-13:
thanks for your reply, I will try it later.

linzha0, 2021-05-12:
Hi, I just monitor the pulse in oscilloscope, actually the pulse looks right and one period(theoretical 1s) is about 999.5ms, the accuracy is less 1ms. I think the previously issue probably is from delayMicroseconds(10000), that makes system timer, which use for nh->now(), not stable. That's why I can see ros time not right.

As suggested from Arduino Forum, if I use timer configured in match PWM mode (MPWM) to send pulse. Some code shown below, at least the nh->now() looks right.
```
#define USE_USBCON

#include <ros.h>
#include <std_msgs/Header.h>

ros::NodeHandle nh;
std_msgs::Header header;
ros::Publisher timer("timer", &header);
volatile boolean isRead = false;

void setup() 
{
  nh.initNode();
  nh.advertise(timer);

/** Set timer TCC2 generate a 100ms pulse every second on D13 (PA17) **/
  GCLK->CLKCTRL.reg = GCLK_CLKCTRL_CLKEN |       // Connect GCLK0 at 48MHz as a clock source for TCC2 and TC3
                      GCLK_CLKCTRL_GEN_GCLK0 |             
                      GCLK_CLKCTRL_ID_TCC2_TC3;            

  // Enable D13's peripheral multiplexer
  PORT->Group[g_APinDescription[13].ulPort].PINCFG[g_APinDescription[13].ulPin].bit.PMUXEN = 1;
  // Set D13 multiplexer switch to position
  PORT->Group[g_APinDescription[13].ulPort].PMUX[g_APinDescription[13].ulPin >> 1].reg |= PORT_PMUX_PMUXO_E;
  
  TCC2->WAVE.reg = TCC_WAVE_WAVEGEN_NPWM;        // Configure the TCC2 timer for normal PWM mode
  while (TCC2->SYNCBUSY.bit.WAVE);               // Wait for synchronization
  
  TCC2->PER.reg = 46874;                         // Set the TCC2 PER register to generate a 1 second period
  while (TCC2->SYNCBUSY.bit.PER);                // Wait for synchronization
  TCC2->CC[1].reg = 4688;                        // Set the TCC2 CC1 register to generate a duty cycle of 10% (100ms)
  while (TCC2->SYNCBUSY.bit.CC1);                // Wait for synchronization

  NVIC_SetPriority(TCC2_IRQn, 0);    // Set the Nested Vector Interrupt Controller (NVIC) priority for TCC2 to 0 (highest)
  NVIC_EnableIRQ(TCC2_IRQn);         // Connect TCC2 to Nested Vector Interrupt Controller (NVIC)

  TCC2->INTENSET.reg = TCC_INTENSET_OVF;       // Enable TCC2 overflow (OVF) interrupts
 
  TCC2->CTRLA.reg = TCC_CTRLA_PRESCSYNC_PRESC |   // Reset timer on the next prescaler clock
                    TCC_CTRLA_PRESCALER_DIV1024;  // Set prescaler to 1024, 48MHz/1024 = 46875kHz                       
  
  TCC2->CTRLA.bit.ENABLE = 1;                    // Enable the TCC2 timer
  while (TCC2->SYNCBUSY.bit.ENABLE);             // Wait for synchronization
}

void loop() {
  while(!isRead);                                 // Block until the trigger() function returns
  isRead = false;                                 // Reset the count read flag
  timer.publish( &header );

  nh.spinOnce();
}

void TCC2_Handler()                                         // Interrupt Service Routine (ISR) for timer TCC2
{      
  if (TCC2->INTFLAG.bit.OVF && TCC2->INTENSET.bit.OVF)      // Optionally check for overflow (OVF) interrupt      
  {   
    TCC2->INTFLAG.bit.OVF = 1;                              // Clear the overflow (OVF) interrupt flag
    // Add optional ISR code here...

    header.stamp = nh.now();
    isRead = true;

  }

}

void TC3_Handler()                                           // Interrupt Service Routine (ISR) for timer TC3
{      
  if (TC3->COUNT16.INTFLAG.bit.OVF && TC3->COUNT16.INTENSET.bit.OVF)       // Optionally check for overflow (OVF) interrupt      
  {   
    // Add optional ISR code here...
    // Put your timer overflow (OVF) code here:     
    header.stamp = nh.now();
    isRead = true;

    REG_TC3_INTFLAG = TC_INTFLAG_OVF;         // Clear the OVF interrupt flag
  }
}
   
```

problem solved, close this issue now and thanks for the help.

closed 2021-05-12 by linzha0