ethz-asl/versavis #28: /home/yuan/Downloads/versavis-master/firmware/libraries/versavis/src/Sensor.h:19:10: fatal error: versavis/ImuMicro.h: No such file or directory

zhangyuan12121, 2021-04-26:
Hello, author, when I use the arduino IDE to compile the firmware, the IDE reports an error, suggesting that the ImuMicro.h file is missing.

floriantschopp, 2021-04-27:
Typically, this appears when you either don't have the correct settings to look for libraries described [here](https://github.com/ethz-asl/versavis#flash-firmware-on-the-versavis-board) or the libraries are not installed which happens with the `setup.sh` script as described [here](https://github.com/ethz-asl/versavis#clone-and-build).

zhangyuan12121, 2021-04-28:
Thanks @**floriantschopp**  for helping me figuring out the problem!
An error occurred when executing the ./setup.sh file,
./setup.sh: line 7: /home/yuan/VersaVIS_ws/devel/setup.: No such file or directory

Modify the fourth line of code in the setup.sh file to
source_path="$devel"/setup.bash
This problem is solved

floriantschopp, 2021-04-28:
OK, thanks for reporting the problem!

closed 2021-04-28 by floriantschopp